memfis19/Annca #7: onActiviyResult

shiv71, 2016-12-22:
when i call this code by fragment 

AnncaConfiguration.Builder videoLimited = new AnncaConfiguration.Builder(activity, CAPTURE_MEDIA);
               videoLimited.setMediaAction(AnncaConfiguration.MEDIA_ACTION_VIDEO);
               videoLimited.setMediaQuality(AnncaConfiguration.MEDIA_QUALITY_AUTO);
               videoLimited.setVideoFileSize(5 * 1024 * 1024);
               videoLimited.setMinimumVideoDuration(1 * 30 * 1000);
               videoLimited.setVideoDuration(1 * 30 * 1000);

               new Annca(videoLimited.build()).launchCamera();

onActivityResult method not calling in fragment

memfis19, 2016-12-22:
Hi, thank you for your feedback. Indeed, it's fault in code, will be fixed asap.

shiv71, 2016-12-22:
if you get any solution regarding this issue than please reply here asap because it is urgent for me 
Thanks

memfis19, 2016-12-22:
The issue connected with how camera activity launches, now I launch activity next way: ```activity.startActivityFoResult```, so in this case fragment will not receive callback in ```onActivityResult```. I will fix it. But if you really need to fix this issue very fast, you can use next workarounds:
1. **Callback**, add some custom callback to fragment and invoke it in fragment, when activity, which hosts fragment, will receive result in ```onActivityResult```
2. Force invoke ```onActivityResult``` in fragment. You can do it with this code in activity which hosts your fragment:
```
@Override
    protected void onActivityResult(int requestCode, int resultCode, Intent data) {
        super.onActivityResult(requestCode, resultCode, data);
        List<Fragment> fragments = getSupportFragmentManager().getFragments();
        if (fragments != null) {
            for (Fragment fragment : fragments) {
                if (fragment != null) fragment.onActivityResult(requestCode, resultCode, data);
            }
        }
    }
```
In case you're using support fragments in other case replace ```getSupportFragmentManager``` with ```getFragmentManager```.

memfis19, 2016-12-22:
Fixed in version 0.3.2. Please look at sample project.

closed 2016-12-22 by memfis19

shiv71, 2016-12-28:
what is new in 0.3.2 version.

memfis19, 2016-12-28:
Hi, added next features:
-Flash supporting;
-Launching camera from fragments, to receive ```onActivityResult``` within fragment.

shiv71, 2017-01-02:
hello memfis19 
Flash button not show on video recording page it show only photo capture page