memfis19/Annca #8: Error capturing video

aresares, 2016-12-27:
Hi, I'm trying to record video but app crash.
This is the log:
```
12-27 17:32:01.052 6327-6327/io.github.memfis19.sample W/System: ClassLoader referenced unknown path: /data/app/io.github.memfis19.sample-1/lib/x86_64
12-27 17:32:01.070 6327-6327/io.github.memfis19.sample I/InstantRun: Instant Run Runtime started. Android package is io.github.memfis19.sample, real application class is null.
                                                                     
                                                                     [ 12-27 17:32:01.072  1541: 1562 D/         ]
                                                                     HostConnection::get() New Host Connection established 0x7bdee161e960, tid 1562
12-27 17:32:01.532 6327-6327/io.github.memfis19.sample W/System: ClassLoader referenced unknown path: /data/app/io.github.memfis19.sample-1/lib/x86_64
12-27 17:32:01.801 6327-6327/io.github.memfis19.sample W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
12-27 17:32:02.335 6327-6355/io.github.memfis19.sample I/OpenGLRenderer: Initialized EGL, version 1.4
12-27 17:32:02.335 6327-6355/io.github.memfis19.sample D/OpenGLRenderer: Swap behavior 1
12-27 17:32:02.477 6327-6355/io.github.memfis19.sample E/EGL_emulation: tid 6355: eglSurfaceAttrib(1174): error 0x3009 (EGL_BAD_MATCH)
12-27 17:32:02.477 6327-6355/io.github.memfis19.sample W/OpenGLRenderer: Failed to set EGL_SWAP_BEHAVIOR on surface 0x7bdee667f100, error=EGL_BAD_MATCH
12-27 17:32:04.360 6327-6327/io.github.memfis19.sample I/CameraManagerGlobal: Connecting to camera service
12-27 17:32:04.517 6327-6355/io.github.memfis19.sample E/EGL_emulation: tid 6355: eglSurfaceAttrib(1174): error 0x3009 (EGL_BAD_MATCH)
12-27 17:32:04.517 6327-6355/io.github.memfis19.sample W/OpenGLRenderer: Failed to set EGL_SWAP_BEHAVIOR on surface 0x7bdee667f280, error=EGL_BAD_MATCH
12-27 17:32:04.632 6327-6355/io.github.memfis19.sample E/EGL_emulation: tid 6355: eglSurfaceAttrib(1174): error 0x3009 (EGL_BAD_MATCH)
12-27 17:32:04.632 6327-6355/io.github.memfis19.sample W/OpenGLRenderer: Failed to set EGL_SWAP_BEHAVIOR on surface 0x7bdee667f280, error=EGL_BAD_MATCH
12-27 17:32:04.736 6327-6355/io.github.memfis19.sample D/OpenGLRenderer: endAllActiveAnimators on 0x7bdee66bf400 (RippleDrawable) with handle 0x7bdee5b2e0a0
12-27 17:32:08.550 6327-6453/io.github.memfis19.sample E/WVMExtractor: Failed to open libwvm.so: dlopen failed: library "libwvm.so" not found
12-27 17:32:08.555 6327-6454/io.github.memfis19.sample I/OMXClient: MuxOMX ctor
12-27 17:32:08.579 6327-6387/io.github.memfis19.sample I/MediaRecorderJNI: prepare: surface=0x7bdefd853e00
12-27 17:32:08.598 6327-6387/io.github.memfis19.sample E/MediaRecorder: start failed: -2147483648
12-27 17:32:08.599 6327-6387/io.github.memfis19.sample E/AndroidRuntime: FATAL EXCEPTION: BaseCameraManager
                                                                         Process: io.github.memfis19.sample, PID: 6327
                                                                         java.lang.RuntimeException: start failed.
                                                                             at android.media.MediaRecorder.start(Native Method)
                                                                             at io.github.memfis19.annca.internal.manager.impl.Camera1Manager$4.run(Camera1Manager.java:143)
                                                                             at android.os.Handler.handleCallback(Handler.java:751)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                             at android.os.Looper.loop(Looper.java:154)
                                                                             at android.os.HandlerThread.run(HandlerThread.java:61)
12-27 17:32:08.614 6327-6464/io.github.memfis19.sample I/OMXClient: MuxOMX ctor
12-27 17:32:08.650 6327-6355/io.github.memfis19.sample E/EGL_emulation: tid 6355: eglSurfaceAttrib(1174): error 0x3009 (EGL_BAD_MATCH)
12-27 17:32:08.650 6327-6355/io.github.memfis19.sample W/OpenGLRenderer: Failed to set EGL_SWAP_BEHAVIOR on surface 0x7bdee667f2c0, error=EGL_BAD_MATCH
12-27 17:32:08.716 6327-6327/io.github.memfis19.sample W/AudioTrack: AUDIO_OUTPUT_FLAG_FAST denied by server; frameCount 16808
```

memfis19, 2016-12-28:
Hi, I will take a look asap. Can you please send your device, android version details.

aresares, 2016-12-29:
I'm using an emulator

`

Name: 5.4_FWVGA_API_25_Android_7.1.1

CPU/ABI: Google APIs Intel Atom (x86_64)

Path: C:\Users\Ares\.android\avd\5.4_FWVGA_API_25_Android_7.1.1.avd

Target: google_apis [Google APIs] (API level 25)

Skin: 480x854

SD Card: C:\Users\Ares\.android\avd\5.4_FWVGA_API_25_Android_7.1.1.avd\sdcard.img

hw.dPad: no

runtime.network.speed: full

hw.accelerometer: yes

hw.device.name: 5.4in FWVGA

vm.heapSize: 16

skin.dynamic: yes

hw.device.manufacturer: Generic

hw.gps: yes

hw.initialOrientation: Portrait

skin.path.backup: _no_skin

image.androidVersion.api: 25

hw.audioInput: yes

image.sysdir.1: system-images\android-25\google_apis\x86_64\

tag.id: google_apis

showDeviceFrame: no

hw.camera.back: emulated

hw.mainKeys: yes

AvdId: 5.4_FWVGA_API_25_Android_7.1.1

hw.camera.front: emulated

hw.lcd.density: 160

avd.ini.displayname: 5.4  FWVGA API 25 Android 7.1.1

hw.gpu.mode: auto

hw.device.hash2: MD5:fbd5143f5b48ba972391c87c302c0c69

hw.ramSize: 512

hw.trackBall: no

hw.battery: yes

hw.cpu.ncore: 2

hw.sdCard: yes

tag.display: Google APIs

runtime.network.latency: none

hw.keyboard: yes

hw.sensors.proximity: yes

disk.dataPartition.size: 800M

hw.sensors.orientation: yes

avd.ini.encoding: UTF-8

hw.gpu.enabled: yes`

memfis19, 2017-01-13:
Hi, sorry for long response. But the answer is quite easy: **"Note: Currently, MediaRecorder does not work on the emulator."**: https://developer.android.com/reference/android/media/MediaRecorder.html

closed 2017-01-13 by memfis19