memfis19/Annca #10: add autofocus on view on click -- require camera instance ,but cant find where to start

madibalive, 2017-01-05:
hi awesome library , but current autofocus take too long on samsung tested on s6 , tried adding the autofocus on code using a a viewOnclickListener , on the main baseAnncaActivity , but it i can grab Camera instance , and also i realise it uses  surfaceview and base camera for camera2,camera1 currently this code is for camera1 , where i assume Camera instance is available 

code is as follows 
```
private Camera mCamera;
mIsAutoFocusing =false; // can be used as a flag , autofocus is already implemented
 
  @Override // onviewClicklistener
    public void onClick(View view) {
            if (mCamera == null || mIsAutoFocusing) return;
            try {
                mIsAutoFocusing = true;
                mCamera.cancelAutoFocus();
                mCamera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override
                    public void onAutoFocus(boolean success, Camera camera) {
                        mIsAutoFocusing = false;
                        if (!success)
                            Toast.makeText(getActivity(), "Unable to auto-focus!", Toast.LENGTH_SHORT).show();
                    }
                });
            } catch (Throwable t) {
                t.printStackTrace();
            }
        } else {
            super.onClick(view);
        }
    }
```

memfis19, 2017-01-05:
Hi, thank you for your feedback, I will try to add, some tools to handle it asap, hope next week will publish next version.

memfis19, 2017-01-13:
Hi, seems slow focus connected to specific device, it is possible to solve but can affect another functionality. How to fix it, you should create your custom camera activity extended from ```AnncaCameraActivity```, and in version ```0.3.3``` you'll be able to customize your camera parameters with the help of next methods ```boolean handleParameters(ParametersHandler<T> parameters);``` or ```void handleCamera(CameraHandler<T> cameraHandler);```, here is some examples:
## **1. For camera 1 api:**
```
@Override
    protected void onCameraControllerReady() {
        super.onCameraControllerReady();

        getCameraController().getCameraManager().handleParameters(new ParametersHandler<Camera.Parameters>() {
            @Override
            public Camera.Parameters getParameters(Camera.Parameters cameraParams) {
                cameraParams.setColorEffect(Camera.Parameters.EFFECT_SEPIA);
                return cameraParams;
            }
        });
    }
```
## **2. For camera 2 api:**
```
    @Override
    protected void onCameraControllerReady() {
        super.onCameraControllerReady();

        getCameraController().getCameraManager().handleParameters(new ParametersHandler<CaptureRequest.Builder>() {
            @Override
            public CaptureRequest.Builder getParameters(CaptureRequest.Builder cameraParams) {
                cameraParams.set(CaptureRequest.CONTROL_EFFECT_MODE, CaptureRequest.CONTROL_EFFECT_MODE_SEPIA);
                return cameraParams;
            }
        });
    }
```
In current examples used color effects, but you can set your focus, flash etc. parameters.

closed 2017-01-13 by memfis19

madibalive, 2017-01-20:
i added ontouch to listen for screen touch then added a extra method on the `BaseCameraManger`  

for camera1manger implementation
```onReFocus (){
       if (camera== null || mIsAutoFocusing) return; // introduce a local variable to camera manger class
            try {
                mIsAutoFocusing = true;
                camera.cancelAutoFocus();
                camera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override
                    public void onAutoFocus(boolean success, Camera camera) {
                        mIsAutoFocusing = false;
                        if (!success)
                            Toast.makeText(getActivity(), "Unable to auto-focus!", Toast.LENGTH_SHORT).show();
                    }
                });
            } catch (Throwable t) {
                t.printStackTrace();
            }
        } else {
            super.onClick(view);
        }
}```

for the autofocusing write before take picture is called , to make all picture clear ,same manger class

PS-- I THINK YOU SHOULD ADD , IMAGES AT NIGHT AT TERRIBLE WITH THE CURRENT IMPLEMENTATION

````
 @Override
    public void takePhoto(File photoFile, CameraPhotoListener cameraPhotoListener) {
        this.outputPath = photoFile;
        this.photoListener = cameraPhotoListener;
        backgroundHandler.post(new Runnable() {
            @Override
            public void run() {
                setCameraPhotoQuality(camera);
                camera.cancelAutoFocus();
                camera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override // didnt check for boolean as i already have manual autofocus , but can check is            
                                      //  autofocus is in call                         mIsAutoFocusing = false;

                    public void onAutoFocus(boolean b, Camera camera) {
                        camera.takePicture(null, null, currentInstance); 
                    }
                });
            }
        });
    }