memfis19/Annca #11: Flashmode on automatic is flashes once ,instead of two

madibalive, 2017-01-05:
taking photo at night is very painfull as the flash light in automode flashes once , initial for the focusing the view , but not when the actual camera takePhoto is called , thus require a second flashes , i realise it pass to android, to take care of this , but help would be appreciated .

memfis19, 2017-01-05:
Hi, thank you again for feedback, seems I miss such case, will be fixed, but need some time to find right solution for this for both camera apis. If you have some ideas, please write here.

memfis19, 2017-01-13:
Hi, I added new methods in version 0.3.3 to make you able to customize your camera parameters with the help of next methods ```boolean handleParameters(ParametersHandler<T> parameters);``` or ```void handleCamera(CameraHandler<T> cameraHandler);```.  But it will be possible only in your custom CameraActivity. You can take a look and ```BaseAnncaActivity```.

closed 2017-01-13 by memfis19

madibalive, 2017-01-20:
Hi,  the flash issue is non existent as it was coming from one line in mediaActionSwitch , on Video mode  u set the flash to invisible , for recording video , i guess it choice since if you want flash on video or not , so i didnt try to git commit it with a solution 
```
 public void setMediaActionState(@MediaActionSwitchView.MediaActionState int actionState) {
        if (mediaActionState == actionState) return;
        if (MediaActionSwitchView.ACTION_PHOTO == actionState) {
            recordButton.setMediaAction(AnncaConfiguration.MEDIA_ACTION_PHOTO);
            if (hasFlash)
                flashSwitch.setVisibility(VISIBLE); // 
        } else {
            recordButton.setMediaAction(AnncaConfiguration.MEDIA_ACTION_VIDEO);
            flashSwitch.setVisibility(GONE); // set invisible here 
        }
        mediaActionState = actionState;
        mediaActionSwitchView.setMediaActionState(actionState);
    }

```

now for the single flashmode , see my other issue on the autofocus , i just added set flashmode to torch and before call to autofocus, and switch off , ofcourse i check if flashmode, is auto first 

Thanks .