memfis19/Annca #12: Provide custom locales

AlexBrechmann, 2017-01-24:
Hi there,
first - thanks for providing this library to everyone it'll save me a lot of time.
 
I am using your library and currently it provides english and I assume danish locales only. Is there any way to change/add locales e.g. for the different textViews within `activity_preview` or for the dialog you create within `onSettingsClick(...)`? 
Currently I get your library via Gradle and actually I don't want to import it manually and modify it myself.

It would be nice to hear from you.
Alex

memfis19, 2017-01-24:
Hi, thank you for your feedback. Currently I am working on next version and I will think how to add this. As quick solution without a lot of modifying you can just put the required locale strings into your application project **res/values** but use the [same names](https://github.com/memfis19/Annca/blob/master/annca/src/main/res/values/strings.xml) as in library for them and they will be override by your application values. As for next version release date - it will be beginning of the February.

AlexBrechmann, 2017-01-24:
Thanks for your fast response. For now I'll got with your provided solution.

AFAP, 2017-02-21:
thank you.谢谢作者

kcn1-71, 2017-03-16:
Hi there, thanks alot for your work =) 
I am newbie, is there any way to setPreviewCallback ?