memfis19/Annca #13: Flash is now showing in Video mode

sanurag51, 2017-01-31:
Hello,

Thanks for providing this awesome code, 
Sir, I need flash in video mode. In this code flash is only enable in camera, but when i switch to the videoCapture mode flash icon is not visible. Please guide me or provide a code snippet to enable flash while capturing video.

Thanks

memfis19, 2017-01-31:
Hi, in current version with out of box solution its not possible. You can implement our own control view - see description how to do it, of just fork the lib and make your changes. In nearest future, this or next week I will release new version and I'll try to add such configuration possibility.

sanurag51, 2017-01-31:
ok sir thanks for response. i will try by my own hope for the best.

closed 2017-01-31 by sanurag51

roys88, 2018-01-23:
Hi memfis19,

We are using your library but are facing the same issue as faced by some other user above.
We also tried to add custom flash to the library but that does not work.

Have there been any new updates regarding it?

Thanks!

meetvishvajit, 2018-01-23:
Hi memfis19,

We are using your library but are facing the same issue as faced by some other user above.
We also tried to add custom flash to the library but that does not work.

Have there been any new updates regarding it?

Thanks!