memfis19/Annca #14: Foucs on touch

dhruv2601, 2017-02-01:
Awesome time working on your library, great work really.
But it would be so much useful if a feature of focus on touch is added in the library.
I tried myself but could not get a way around to successfully implement the feature.

memfis19, 2017-02-01:
Hi, thank you for your feedback. I've already received such request and I'm working on it. But anyway I want to clarify  how have you used lib? (I mean you are using out of box solution or implementing custom like described in manual).

memfis19, 2017-02-01:
In case you're implementing custom you can try next steps:
1. `previewContainer` where you can register `onTouch` or `onClick` listener;
2. `getCameraController().getCameraManager().handleParameters(ParametersHandler<CameraParameters> parameters);` where `CameraParameters` is type `Camera.Parameters` for Camera1 and `CaptureRequest.Builder` for Camera2 - with this method you can update camera parameters;
3. `getCameraController().getCameraManager().handleCamera(CameraHandler<Camera> cameraHandler);`, where `Camera` is type of `Camera` for Camera1 and `CameraDevice` for Camera2 - with this method you can get reference on camera instance.
You should call this method when camera is ready, i.e. in `onCameraReady` method.
And then implement focus on touch. i.e. for camera 1:
```
public void onClick(View view) {
                camera.cancelAutoFocus();
                camera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override
                    public void onAutoFocus(boolean success, Camera camera) {
                      //your code
                    }
                });
    }
```

dhruv2601, 2017-02-03:
I have actually combined one more library with Annca, as I needed a really good image cropper too.
I will surely get into the code you have provided :)

closed 2017-02-20 by memfis19