memfis19/Annca #15: How to zoom?

smaznet, 2017-02-06:
How to change zoom of the camera?

memfis19, 2017-02-06:
Hi, it doesn't work in current version within Annca default solution (will be added soon). But you can still do it, if don't want to wait, with custom approach described in readme where you will have access to the camera parameters and set zoom level.

smaznet, 2017-02-06:
@memfis19
thank you  another question how I can restart review after photo taking
where should I make a method for it?

memfis19, 2017-02-07:
Hi, can you please describe your use case? As I know after you are taking photo, preview activity is launching and you can retake photo from preview screen. Or do you want to take series of photos?

memfis19, 2017-02-07:
In case you are using default solution - you have repeat button on preview screen. In case you are using custom implementation - just don't close your activity on `onPhotoTaken` method. I've just pushed new example of creating custom camera, please look at `CustomCameraActivity` in example project.

smaznet, 2017-02-07:
i want use your camera api in my app for making a simple camera app

memfis19, 2017-02-07:
Then I assume you need to make your custom implementation. Please look at example in `CustomCameraActivity` in sample project. For adding zoom you can implement your own controls and use `getCameraController().getCameraManager().handleParameters(ParametersHandler<CameraParameters> parameters);` method where you can put your zoom settings. I.e.:

```
zoomButton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View view) {
                getCameraController().getCameraManager().handleParameters(new ParametersHandler<Camera.Parameters>() {
                    @Override
                    public Camera.Parameters getParameters(Camera.Parameters parameters) {
                        parameters.setZoom(zoom);
                        return parameters;
                    }
                });
            }
        });
```

smaznet, 2017-02-07:
@memfis19 thank you for zoom

where is the code of camera that pause preview after taken?

memfis19, 2017-02-07:
What do mean by pause the preview? You can take the photo and show it on preview screen. In current implementation camera lifecycle is bind to the activity lifecycle, when you leave the activity then `CameraController` method `onPause` is called and `onDestroy` respectively.

closed 2017-02-20 by memfis19