memfis19/Annca #16: How to change outfile path?

AFAP, 2017-03-16:
HI,can we change the path of photo?I see the method 'getoutputfile' is private.How to change the target floder?
Thanks.

memfis19, 2017-03-16:
Hi, to get output file you should use next approach` String filePath = data.getStringExtra(AnncaConfiguration.Arguments.FILE_PATH);` on `onActitvtyResult ` method. It is not possible right now, but I will add it.

AFAP, 2017-03-16:
Thanks.
Now,I can copy photo to target folder(Use "data.getStringExtra(AnncaConfiguration.Arguments.FILE_PATH)" on onActitvtyResult),but ,if we can set the target folder by configuration, that's so good.

Mehul1000, 2017-05-29:
Hi memfis19,

Thanks for providing good example of custom camera. Can you please update code where i can set my own path with different folder? Also, in CustomCamera2Activity and CustomCameraActivity can you write code for flash (Auto,ON,OFF) and switch camera (Front/Back)?

memfis19, 2017-05-29:
Hi, yes I'll add sample for flash and camera switch soon. As for customizing path I'am not sure about it, but will try to do it asap also. Anyway you can use approach provided by @AFAP and copy from specified path to your own. (Currently I'm working on quite big update for library so I'm not sure about adding custom path feature in current version).