memfis19/Annca #20: how to maintain the ratio of video image?

ivandjava, 2017-04-06:
Hi, I apologize in advance if my language is not good :)

https://drive.google.com/open?id=0B44cB1gVSzPLT3lJUnVhSFMzd3c

I'm making a video camera and try to record a video. The problem is every time I change the orientation of the smartphone, which previewed image does not have the same image ratio. 
As Figures 1 and 2:
[image]
[image]

Please help me to improve my video  :'(

memfis19, 2017-04-06:
Hi, 
As i can see you do not lock device orientation. How do you handle device rotation?

ivandjava, 2017-04-07:
I mean the question, how to maintain ratio of the image even if the smartphone is rotated?
What's wrong with my program?

memfis19, 2017-04-07:
1. You do not keep aspect ratio for the surface view. Check you surface view aspect ratio and your preview size aspect ratio, they should much, i.e.: `yourView.getHeight()/yourView.getWidth()` to `previewSize.height/previewSize.width`.
2. Do not use ```mCamera.setDisplayOrientation``` hardcoded in case you are allow rotating screen. If you support it then you need to recalculate display orientation degrees. I.e.
```
public int getRotation(Context context) {
            final int rotation = ((WindowManager) context.getSystemService(Context.WINDOW_SERVICE)).getDefaultDisplay().getOrientation();
            switch (rotation) {
                case Surface.ROTATION_0:
                    return 90;
                case Surface.ROTATION_90:
                    return 0;
                case Surface.ROTATION_180:
                    return 90;
                default:
                    return 0;
            }
        }
```
P.S.: I'll close this issue. **It's not relevant to this library** in case you have such questions then direct mail me. Email presented in my profile.

closed 2017-04-07 by memfis19