memfis19/Annca #23: Custom layout: sample app crashes

yargeel, 2017-05-04:
Hello!
Sample app crashes when you tap "Take photo" on custom layout again:
Process: io.github.memfis19.sample, PID: 14415
java.lang.RuntimeException: takePicture failed
at android.hardware.Camera.native_takePicture(Native Method)
at android.hardware.Camera.takePicture(Camera.java:1833)
at android.hardware.Camera.takePicture(Camera.java:1778)
at io.github.memfis19.annca.internal.manager.impl.Camera1Manager$3.run(Camera1Manager.java:128)
at android.os.Handler.handleCallback(Handler.java:815)
at android.os.Handler.dispatchMessage(Handler.java:104)
at android.os.Looper.loop(Looper.java:207)
at android.os.HandlerThread.run(HandlerThread.java:61)

memfis19, 2017-05-04:
Hi, what device do you use?

yargeel, 2017-05-04:
Hello! 
Tested on genimotion emu (4.4.4), ZTE a510 (6.0), Sony xperia z1 (5.1.1) and sams galaxy Tab s2 (6.0)

memfis19, 2017-05-29:
Hi,
Spend some time investigating it using Genimotion and Galaxy Tab s2 , but can't reproduce it.

closed 2017-05-29 by memfis19