memfis19/Annca #27: Directly start video recorder

hardikJoshi123, 2017-05-29:
Hello, 

 its a great library to record videos. I want to start video recorder directly when user click on buttton, I want to open full screen view which will record video, user can stop video by pressing stop button. Can you suggest if i can do it with this library ?

memfis19, 2017-05-29:
Hi, do you want to press start button on your own activity?

hardikJoshi123, 2017-05-29:
No. 

new Annca(videoLimited.build()).launchCamera();

 This will open a new activity having setting, camera flip and record button. I want to start recording directly when this activity will open. Also, i want to hide settings and flip buttons. User can only stop video recording.

memfis19, 2017-07-13:
Hi, this is impossible with out of box solution, bit it is possible with customizing approach: you can check example how to customize and then i.e. just pass some additional parameter within bundle which will tells activity to start recording immediately.

closed 2017-07-13 by memfis19

Anu07, 2021-01-06:
Hi

@hardikJoshi123 Were you able to get it working?