memfis19/Annca #29: Hi!How to differentiate type of action within onActivityResult?

bakhman1981, 2017-07-10:
HI! 
this is not issue, i think it already exist.
How can I understand  Photo or Video was captured?

memfis19, 2017-07-10:
Hi, well depends how do you use it. I suppose you are using not custom approach, so **_Annca_** does not provide any info in `onActivityResult`, because it is quite easy to retrieve file extension or mime-type with standart tools i.e. `File` (in case you really need it - I'll add, but not in nearest time, sorry). As for settings, sure you pass general setting both for photo and video actions. In case something does not work - can you send me please how do you set settings.

bakhman1981, 2017-07-11:
Hi!Yes- it works great about video restrictions! Only one question - it is strange and seems like media quality parameter just retrieve available values from Camera resolution config. is it correct? because when I select 'LOW' for 15sec = 244Kb size. - when I select 'Medium' for 15 sec = 22Mb sizethis is really big jump between low and medium)) and in your config.java i see that LOW=11 and MEDIUM=12. - so no ability to select quality something really medium like 1mb size. And if I selectsetVideoFileSize(1024*1024) system anyway as i understand selects LOW mode and final file size not 1mb as I set, but the same 244Kb(((

bakhman1981, 2017-07-11:
ACtually there is some bug - i don't know how to solve it:1. I've checked that yes - your Annca takes available resolution modes from camera option.2. BUt what strange - in native camera app - it shows 4 modes (144x144, 640x480, 1280, 1940). in annca camera mode it shows (144x144, 1280, 1940).3. So native camera has 4 modes, but annca only 3 modes. how to solve this?(( 11.07.2017, 07:51, "zhjcnm@yandex.ru" <zhjcnm@yandex.ru>:Hi!Yes- it works great about video restrictions! Only one question - it is strange and seems like media quality parameter just retrieve available values from Camera resolution config. is it correct? because when I select 'LOW' for 15sec = 244Kb size. - when I select 'Medium' for 15 sec = 22Mb sizethis is really big jump between low and medium)) and in your config.java i see that LOW=11 and MEDIUM=12. - so no ability to select quality something really medium like 1mb size. And if I selectsetVideoFileSize(1024*1024) system anyway as i understand selects LOW mode and final file size not 1mb as I set, but the same 244Kb(((

memfis19, 2017-07-13:
Hi, well yes, it is thing to improve and I believe I mentioned it in README. I don't know, do you know , but device camera has predefined amount of supported sizes for preview, photo and video (and they are different on different devices). So in **Annca** was selected more simple approach for end using - `MEDIA_QUALITY`.  And it has few values: **low, medium, high**. Depends on video sizes this `MEDIA_QUALITY` is evaluated. Algorithm is very simple just split the supported sizes on `MEDIA_QUALITY` amounts. I.e. you have next supported values: **144x144, 256x372, 320x480, 480x640, 480x720, 720x1280, 1080x1920**. So **Annca** will use _144 for low_, _1080 for hight_ and _480 for medium_. Beside this to start record we need provide additional setting such as _audio and video bitrate, codec_ etc. This values are retrieved from [`CamcorderProfile`](https://developer.android.com/reference/android/media/CamcorderProfile.html) profiles and matched to the `MEDIA_QUALITY` sizes **(note that `CamcorderProfile` may return profile which is not supported by camera, so it is search for best matching option)**, and then used all this values. That's why the difference between **LOW** and **MEDUIM** quality so big, 144 audio and video bitrates very low and for 480p quite big. _Exceptional case is limitation by file size_ in this case audio and video bitrates are decreasing until video will fit file size or will be used the lowest possible values that camera supports.

bakhman1981, 2017-08-05:
Hello! Can you little bit help me : what is the most correct way/solution to implement 'imagepicker' from gallery within your Annca library? Thanks a lot! Of course I can just make separate button on screeen outside Annca call to camera, but more great way to add this when camera opened from Annca call.