memfis19/Annca #51: Video is not recording in android 10+

kranthi0987, 2020-09-02:
**E/CameraHelper: Couldn't find any suitable preview size**
2020-09-03 01:21:36.533 6554-6554/com.wewill.steppie I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2020-09-03 01:21:36.560 6554-6554/com.wewill.steppie D/InputTransport: Input channel destroyed: 'ClientS', fd=182
2020-09-03 01:21:36.562 6554-6554/com.wewill.steppie D/ViewRootImpl@b579bd5[Camera2Activity]: MSG_RESIZED: frame=(0,54,720,1520) ci=(0,0,0,84) vi=(0,0,0,84) or=1
2020-09-03 01:21:36.581 6554-26668/com.wewill.steppie V/FA: Connection attempt already in progress
2020-09-03 01:21:36.584 6554-26668/com.wewill.steppie V/FA: Connection attempt already in progress
2020-09-03 01:21:36.584 6554-26668/com.wewill.steppie D/FA: Connected to remote service
2020-09-03 01:21:36.585 6554-26668/com.wewill.steppie V/FA: Processing queued up service tasks: 4
2020-09-03 01:21:36.598 6554-6578/com.wewill.steppie I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_OPEN for client com.wewill.steppie API Level 2
2020-09-03 01:21:36.876 6554-6623/com.wewill.steppie W/libEGL: EGLNativeWindowType 0xb6eff088 disconnect failed
2020-09-03 01:21:36.884 6554-6554/com.wewill.steppie D/ViewRootImpl@60f8077[VideoDetails]: Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)8 dur=7 res=0x5 s={false 0} ch=true
2020-09-03 01:21:36.886 6554-6554/com.wewill.steppie D/ViewRootImpl@60f8077[VideoDetails]: stopped(true) old=false
2020-09-03 01:21:37.507 6554-6575/com.wewill.steppie I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client com.wewill.steppie API Level 2
2020-09-03 01:21:41.645 6554-26668/com.wewill.steppie V/FA: Inactivity, disconnecting from the service
2020-09-03 01:21:46.707 6554-6554/com.wewill.steppie D/ViewRootImpl@b579bd5[Camera2Activity]: ViewPostIme pointer 0
2020-09-03 01:21:46.802 6554-6554/com.wewill.steppie D/ViewRootImpl@b579bd5[Camera2Activity]: ViewPostIme pointer 1
2020-09-03 01:21:46.815 6554-6554/com.wewill.steppie **E/SoundPool: error loading /product/media/audio/ui/VideoRecord.ogg**
2020-09-03 01:21:46.817 6554-6554/com.wewill.steppie **E/SoundPool: error loading /product/media/audio/ui/VideoRecord.ogg**
2020-09-03 01:21:46.824 6554-6554/com.wewill.steppie **D/CameraHelper: Failed to create directory.**
2020-09-03 01:21:46.827 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setup
2020-09-03 01:21:46.838 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setAudioSource(5)
2020-09-03 01:21:46.849 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setVideoSource(2)
2020-09-03 01:21:46.849 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setVideoFrameRate(30)
2020-09-03 01:21:46.849 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setVideoSize(720, 480)
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setParameter()
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setVideoEncoder(2)
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setParameter()
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/chatty: uid=10259(com.wewill.steppie) BaseCameraManag identical 1 line
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setParameter()
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: setAudioEncoder(3)
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie **E/Camera2Manager: Error during preparing MediaRecorder: Attempt to invoke virtual method 'java.lang.String java.io.File.toString()' on a null object reference**
2020-09-03 01:21:46.850 6554-26785/com.wewill.steppie I/MediaRecorderJNI: native_reset
2020-09-03 01:21:46.853 6554-26785/com.wewill.steppie I/MediaRecorderJNI: release

MH3sham, 2020-09-09:
@memfis19 Same here (android 10)

MH3sham, 2020-09-09:
@kranthi0987 try adding `android:requestLegacyExternalStorage="true"` in the AndroidManifest.xml file (application tag)

kranthi0987, 2020-09-09:
Thanks @MH3sham  it is working fine now.

closed 2020-09-25 by memfis19